microflo/microflo #72: Cannot Flash MicroFlo runtime to Arduino

pipiHuang, 2015-02-10:
I have some problem about Flash MicroFlo runtime to Arduino.
I follow the tutorial 
https://docs.flowhub.io/getting-started-microflo/

when I follow this tutorial and want to flash MicroFlo runtime to Arduino, 
(https://github.com/microflo/microflo/blob/master/doc/arduino-getstarted.md)
I meet the problem
There are no version 0.4.0 Microflo to download, so I add the version 0.3.2 microflo into Arduino IDE.
After add microflo into arduino IDE, I can't see the example code of microflo.

On the other hand, I follow the readme from this https://github.com/microflo/microflo
when I try this command 
`
make upload GRAPH=examples/blink.fbp MODEL=uno
`
hope to upload the graph, but get some error message.
```
./src/componentlib-source.hpp:1:40: fatal error: core/components/PwmWrite.hpp: No such file or directory
 #include "core/components/PwmWrite.hpp"
                                        ^
compilation terminated.
make[1]: *** [.build/uno/src/main.d] Error 1
make[1]: Leaving directory `/home/pipi/mygit_home/micro2/microflo/build/arduino'
Make failed with code 2
make: *** [build-arduino] Error 1
```
I check the path and find that "core/components/PwmWrite.hpp" really doesn't exist.
I know there are microflo-core package which has PwmWrite.hpp but I don't know where should I put this file.

p.s. I use Arduino uno R3, xubuntu 14 64bit

jonnor, 2015-02-25:
This is a bug in the tarball/release-generation.

jonnor, 2015-02-25:
@pipiHuang It seems that your git version did not automatically. You have to do `git submodule update --init`, try to do `make upload ...` again after that. I have updated the README with this instruction

I cannot reproduce your issue with the Arduino IDE however. Which version of the IDE are you using? The example should be under `File -> Examples -> microflo -> Standalone`

jonnor, 2015-03-09:
Closed, not able to reproduce.

closed 2015-03-09 by jonnor