microflo/microflo #77: arduino-getstarted.md refers to the 0.4.0 release, but it's not at the link

awgrover, 2015-04-11:
The getstarted at
https://github.com/microflo/microflo/blob/master/doc/arduino-getstarted.md
says "Download the latest MicroFlo release. It should be version 0.4.0 or later.",
but the link, to https://github.com/jonnor/microflo/releases, has not such 0.4.0 (latest is 0.3.23)

There is a tag for 0.4.0, but not a "release"

jonnor, 2015-05-13:
Yes, 0.4.0 has not been released due to a blocker issue in Flowhub:
https://github.com/noflo/noflo-ui/issues/363

jonnor, 2017-07-11:
0.4.0 has now been released. However we no longer produce a tarball for release, instead one installs MicroFlo tool via NPM. See https://github.com/microflo/microflo-example-arduino#microflo-example-arduino

closed 2017-07-11 by jonnor