microflo/microflo #84: Get rid of component.io for browser build

jonnor, 2016-02-26:
Should use webpack (or browserify), which uses NPM instead of an obsoleted, parallel universe packaging system.

Should allow us to get rid of the forks we have, like:

* https://github.com/microflo/buffer
* https://github.com/microflo/ieee754
* https://github.com/microflo/base64-js

jonnor, 2016-04-07:
Have had some successes with webpack. One thing is that microflo is pulled into https://www.npmjs.com/package/fbp-protocol-client so that should also be fixed. And that is used by `fbp-spec` and `noflo-runtime`, which again is used by `noflo-ui`.

jonnor, 2017-01-26:
@bergie do we use the component.io version of `fbp-protocol-client` anywhere now? If not then that unblocks this

bergie, 2017-01-26:
@jonnor I don't think we do. Some 3rd parties might but in general component.io is anyway semi-broken by now

jonnor, 2017-07-11:
Done since 0.3.54

closed 2017-07-11 by jonnor