microflo/microflo #87: Error when trying to run the tests

rraallvv, 2016-04-11:
```
Running "mochaTest:nodejs" (mochaTest) task
>> Mocha exploded!
>> Error: Could not locate the bindings file. Tried:
>>  → /Users/user/Desktop/microflo/node_modules/serialport/build/serialport.node
>>  → /Users/user/Desktop/microflo/node_modules/serialport/build/Debug/serialport.node
>>  → /Users/user/Desktop/microflo/node_modules/serialport/build/Release/serialport.node
>>  → /Users/user/Desktop/microflo/node_modules/serialport/out/Debug/serialport.node
>>  → /Users/user/Desktop/microflo/node_modules/serialport/Debug/serialport.node
>>  → /Users/user/Desktop/microflo/node_modules/serialport/out/Release/serialport.node
>>  → /Users/user/Desktop/microflo/node_modules/serialport/Release/serialport.node
>>  → /Users/user/Desktop/microflo/node_modules/serialport/build/default/serialport.node
>>  → /Users/user/Desktop/microflo/node_modules/serialport/compiled/5.10.0/darwin/x64/serialport.node
>>   at bindings (/Users/user/Desktop/microflo/node_modules/bindings/bindings.js:88:9)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/node_modules/serialport/serialport.js:16:44)
>>   at Module._compile (module.js:413:34)
>>   at Object.Module._extensions..js (module.js:422:10)
>>   at Module.load (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:211:36)
>>   at Function.Module._load (module.js:314:12)
>>   at Module.require (module.js:367:17)
>>   at require (internal/module.js:16:19)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/lib/serial.coffee:12:16)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/lib/serial.coffee:1:1)
>>   at Module._compile (module.js:413:34)
>>   at Object.loadFile (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:182:19)
>>   at Module.load (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:211:36)
>>   at Function.Module._load (module.js:314:12)
>>   at Module.require (module.js:367:17)
>>   at require (internal/module.js:16:19)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/lib/runtime.coffee:25:10)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/lib/runtime.coffee:766:4)
>>   at Module._compile (module.js:413:34)
>>   at Object.loadFile (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:182:19)
>>   at Module.load (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:211:36)
>>   at Function.Module._load (module.js:314:12)
>>   at Module.require (module.js:367:17)
>>   at require (internal/module.js:16:19)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/lib/microflo.coffee:4:12)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/lib/microflo.coffee:1:1)
>>   at Module._compile (module.js:413:34)
>>   at Object.loadFile (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:182:19)
>>   at Module.load (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:211:36)
>>   at Function.Module._load (module.js:314:12)
>>   at Module.require (module.js:367:17)
>>   at require (internal/module.js:16:19)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/test/blink.coffee:9:14)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/test/blink.coffee:1:1)
>>   at Module._compile (module.js:413:34)
>>   at Object.loadFile (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:182:19)
>>   at Module.load (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:211:36)
>>   at Function.Module._load (module.js:314:12)
>>   at Module.require (module.js:367:17)
>>   at require (internal/module.js:16:19)
>>   at /Users/user/Desktop/microflo/node_modules/mocha/lib/mocha.js:184:27
>>   at Array.forEach (native)
>>   at Mocha.loadFiles (/Users/user/Desktop/microflo/node_modules/mocha/lib/mocha.js:181:14)
>>   at MochaWrapper.run (/Users/user/Desktop/microflo/node_modules/grunt-mocha-test/tasks/lib/MochaWrapper.js:51:15)
>>   at /Users/user/Desktop/microflo/node_modules/grunt-mocha-test/tasks/mocha-test.js:86:20
>>   at capture (/Users/user/Desktop/microflo/node_modules/grunt-mocha-test/tasks/mocha-test.js:33:5)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/node_modules/grunt-mocha-test/tasks/mocha-test.js:81:5)
>>   at Object.proxiedTaskFn (/Users/user/Desktop/microflo/node_modules/grunt-retro/tasks/retro.js:54:21)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/node_modules/grunt/lib/grunt/task.js:264:15)
>>   at Object.thisTask.fn (/Users/user/Desktop/microflo/node_modules/grunt/lib/grunt/task.js:82:16)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/node_modules/grunt/lib/util/task.js:301:30)
>>   at Task.runTaskFn (/Users/user/Desktop/microflo/node_modules/grunt/lib/util/task.js:251:24)
>>   at Task.<anonymous> (/Users/user/Desktop/microflo/node_modules/grunt/lib/util/task.js:300:12)
>>   at /Users/user/Desktop/microflo/node_modules/grunt/lib/util/task.js:227:11
>>   at _combinedTickCallback (internal/process/next_tick.js:67:7)
>>   at process._tickDomainCallback (internal/process/next_tick.js:122:9)
Warning: Task "mochaTest:nodejs" failed. Use --force to continue.

Aborted due to warnings.
npm ERR! Test failed.  See above for more details.
```

jonnor, 2016-04-17:
Which version are you on? If you are using git, please output results of `git describe --always`

rraallvv, 2016-04-18:
Sorry, I should have posted more details

```
$ git describe --always
0.3.2-349-gbf48fe1
```

System OS X 10.11.4 (El Capitan)
Git version 2.6.4 (Apple Git-63)
Emscripten 1.35.23 installed via homebrew

```
$ git clone —recursive https://github.com/microflo/microflo
$ cd microflo
$ npm install
npm WARN component-coffee@0.1.4 requires a peer of component@>= 0.14 but none was installed.
$ npm test
```

Before I was getting this error

```
...
Running "mochaTest:nodejs" (mochaTest) task
WARN: could not load Emscripten build Error: Cannot find module '../build/emscripten/microflo-runtime.js'
>> Mocha exploded!
>> TypeError: Cannot read property '_emscripten_runtime_new' of undefined
>>   at new RuntimeSimulator (/Users/user/Desktop/microflo/lib/simulator.coffee:55:19)
>>   at Suite.<anonymous> (/Users/user/Desktop/microflo/test/blink.coffee:26:19)
>>   at context.describe.context.context (/Users/user/Desktop/microflo/node_modules/mocha/lib/interfaces/bdd.js:75:10)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/test/blink.coffee:25:1)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/test/blink.coffee:1:1)
>>   at Module._compile (module.js:413:34)
>>   at Object.loadFile (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:182:19)
>>   at Module.load (/Users/user/Desktop/microflo/node_modules/grunt-contrib-coffee/node_modules/coffee-script/lib/coffee-script/coffee-script.js:211:36)
>>   at Function.Module._load (module.js:314:12)
>>   at Module.require (module.js:367:17)
>>   at require (internal/module.js:16:19)
>>   at /Users/user/Desktop/microflo/node_modules/mocha/lib/mocha.js:184:27
>>   at Array.forEach (native)
>>   at Mocha.loadFiles (/Users/user/Desktop/microflo/node_modules/mocha/lib/mocha.js:181:14)
>>   at MochaWrapper.run (/Users/user/Desktop/microflo/node_modules/grunt-mocha-test/tasks/lib/MochaWrapper.js:51:15)
>>   at /Users/user/Desktop/microflo/node_modules/grunt-mocha-test/tasks/mocha-test.js:86:20
>>   at capture (/Users/user/Desktop/microflo/node_modules/grunt-mocha-test/tasks/mocha-test.js:33:5)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/node_modules/grunt-mocha-test/tasks/mocha-test.js:81:5)
>>   at Object.proxiedTaskFn (/Users/user/Desktop/microflo/node_modules/grunt-retro/tasks/retro.js:54:21)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/node_modules/grunt/lib/grunt/task.js:264:15)
>>   at Object.thisTask.fn (/Users/user/Desktop/microflo/node_modules/grunt/lib/grunt/task.js:82:16)
>>   at Object.<anonymous> (/Users/user/Desktop/microflo/node_modules/grunt/lib/util/task.js:301:30)
>>   at Task.runTaskFn (/Users/user/Desktop/microflo/node_modules/grunt/lib/util/task.js:251:24)
>>   at Task.<anonymous> (/Users/user/Desktop/microflo/node_modules/grunt/lib/util/task.js:300:12)
>>   at /Users/user/Desktop/microflo/node_modules/grunt/lib/util/task.js:227:11
>>   at _combinedTickCallback (internal/process/next_tick.js:67:7)
>>   at process._tickDomainCallback (internal/process/next_tick.js:122:9)
Warning: Task "mochaTest:nodejs" failed. Use --force to continue.

Aborted due to warnings.
npm ERR! Test failed.  See above for more details.
```
 
Then copied the files at https://github.com/microflo/microflo-emscripten to build/emscripten in the folder microflo and got this

```
...
  1) a Blink program should load 3 nodes:
     Error: timeout of 2000ms exceeded
    at Timeout.<anonymous> (/Users/user/Desktop/microflo/node_modules/mocha/lib/runnable.js:158:19)
    at tryOnTimeout (timers.js:224:11)
    at Timer.listOnTimeout (timers.js:198:5)
  

  2) a Blink program should go HIGH when timer expires:
     Error: timeout of 5000ms exceeded
    at Timeout.<anonymous> (/Users/user/Desktop/microflo/node_modules/mocha/lib/runnable.js:158:19)
    at tryOnTimeout (timers.js:224:11)
    at Timer.listOnTimeout (timers.js:198:5)
  

  3) a Blink program and then toggle LOW when timer expires again:
     Error: timeout of 5000ms exceeded
    at Timeout.<anonymous> (/Users/user/Desktop/microflo/node_modules/mocha/lib/runnable.js:158:19)
    at tryOnTimeout (timers.js:224:11)
    at Timer.listOnTimeout (timers.js:198:5)
  

  4) a Blink program "after all" hook:
     Error: timeout of 2000ms exceeded
    at Timeout.<anonymous> (/Users/user/Desktop/microflo/node_modules/mocha/lib/runnable.js:158:19)
    at tryOnTimeout (timers.js:224:11)
    at Timer.listOnTimeout (timers.js:198:5)
  

  5) Network Uploading a graph via commandstream gives one response per command:
     Error: timeout of 2000ms exceeded
    at Timeout.<anonymous> (/Users/user/Desktop/microflo/node_modules/mocha/lib/runnable.js:158:19)
    at tryOnTimeout (timers.js:224:11)
    at Timer.listOnTimeout (timers.js:198:5)

Warning: Task "mochaTest:nodejs" failed. Use --force to continue.

Aborted due to warnings.
npm ERR! Test failed.  See above for more details.
```

I don’t have and Arduino or other compatible board so I tried

```
$ make check

rm -rf /Users/user/Desktop/microflo/build/emscripten
mkdir -p /Users/user/Desktop/microflo/build/emscripten
./microflo.js generate examples/blink.fbp /Users/user/Desktop/microflo/build/emscripten --target emscripten 
examples/blink.fbp
cd /Users/user/Desktop/microflo/build/emscripten && emcc -o microflo-runtime.html --pre-js /Users/user/Desktop/microflo/microflo/emscripten-pre.js main.cpp -I. -I/Users/user/Desktop/microflo/microflo -Wall -DMICROFLO_MESSAGE_LIMIT=200 -s NO_DYNAMIC_EXECUTION=1 -s EXPORTED_FUNCTIONS='["_emscripten_runtime_new", "_emscripten_runtime_free", "_emscripten_runtime_run", "_emscripten_runtime_send", "_emscripten_runtime_setup"]' -s RESERVED_FUNCTION_POINTERS=10
error: unknown target triple 'asmjs-unknown-emscripten', please use -triple or -arch
ERROR:root:compiler frontend failed to generate LLVM bitcode, halting
make: *** [build-emscripten] Error 1
```

jonnor, 2016-04-19:
Those tests are currently failing, this is known.

But the `emcc` target triple failure I have not seen before. Which version of Emscripten and which OS?

rraallvv, 2016-04-20:
```
$ emcc --version
emcc (Emscripten gcc/clang-like replacement) 1.35.23 (commit a9c0361a1d83374059fca73485643e89c2772331)
Copyright (C) 2014 the Emscripten authors
```
 Installed via homebrew on Mac OS X 10.11.4 (15E65)

rraallvv, 2016-04-21:
I needed to change `~/.emscripten` like so

```
# this helps projects using emscripten find it
EMSCRIPTEN_ROOT = os.path.expanduser(os.getenv('EMSCRIPTEN') or '/usr/local/Cellar/emscripten/1.35.23/libexec') # directory
LLVM_ROOT = os.path.expanduser(os.getenv('LLVM') or '/usr/local/Cellar/emscripten/1.35.23/libexec/llvm/bin') # directory
```

It seems emscripten is not properly set up by homebrew, with those changes microflow compiles ok

closed 2016-04-21 by rraallvv

jonnor, 2016-04-21:
The tests should noe also have been fixed, passing locally and on Travis CI. Note that there are some new dependencies, need to npm install.

rraallvv, 2016-04-22:
I'll try to build microflo with the new dependencies, thanks for the info.