microflo/microflo #130: Missing timeout on connecting to microcontroller

jonnor, 2017-05-29:
Unclear why it happens, but sometimes runtime does not respond to an "open communication" command. Right now `microflo runtime` code just hangs forever in that case. Should instead time out and abort with a friendly error message.

bergie, 2017-05-29:
Example of such failure:

```
May 19 15:55:39 ingress-table node[287]: MicroFlo runtime listening at 127.0.0.1:4444
May 19 15:55:41 ingress-table node[287]: queuing buf <Buffer 75 43 2f 46 6c 6f 30 31> 0 8 false
May 19 15:55:41 ingress-table node[287]: MICROFLO SEND: 40 <Buffer 75 43 2f 46 6c 6f 30 31> null undefined false
May 19 15:55:41 ingress-table node[287]: MICROFLO Serial: buffered 8 at 8/sec. Time elapsed: 2s. 0 commands in buffer
```

Successful case:

```
May 19 16:00:13 ingress-table node[375]: MicroFlo runtime listening at 127.0.0.1:4444
May 19 16:00:16 ingress-table node[375]: queuing buf <Buffer 75 43 2f 46 6c 6f 30 31> 0 8 false
May 19 16:00:16 ingress-table node[375]: MICROFLO SEND: 40 <Buffer 75 43 2f 46 6c 6f 30 31> null undefined false
May 19 16:00:16 ingress-table node[375]: MICROFLO RECV: 8 <Buffer 13 00 00 00 00 00 00 00>
May 19 16:00:16 ingress-table node[375]: queuing buf <Buffer 0a 00 00 00 00 00 00 00 0f 01 00 00 00 00 00 00 0b 03 00 00 00 00 00 00 0b 17 00 00 00 00 00 00 0b 04 00 00 00 00 00 00 0b 04 00 00 00 00 00 00 0b 17
 ... > 0 336 true
May 19 16:00:16 ingress-table node[375]: FBP MICROFLO RESPONSE IGNORED: communicationopen
May 19 16:00:16 ingress-table node[375]: MICROFLO RECV: 8 <Buffer 73 00 00 00 00 00 00 00>
```

closed 2017-06-01 by jonnor

bergie, 2017-06-02:
Verified live on ingress table:

```
```-- Reboot --
Jun 02 13:35:07 ingress-table systemd[1]: Started microflo-streetlights.service.
Jun 02 13:35:47 ingress-table node[298]: INFO: No runtime id set, generated one: a9403e29-4b78-4980-b110-1f143d1d17a3
Jun 02 13:35:47 ingress-table node[298]: Using serial baudrate with /dev/ttyACM1 115200
Jun 02 13:35:47 ingress-table node[298]: Available serial ports:  [ '/dev/ttyACM0', '/dev/ttyACM1', '/dev/ttyAMA0' ]
Jun 02 13:35:47 ingress-table node[298]: Using port: /dev/ttyACM1
Jun 02 13:35:48 ingress-table node[298]: TimeoutError: Device did not respond within 1000ms
Jun 02 13:35:48 ingress-table node[298]:     at afterTimeout (/opt/ingress-table/node_modules/bluebird/js/release/timers.js:49:15)
Jun 02 13:35:48 ingress-table node[298]:     at Timeout.timeoutTimeout [as _onTimeout] (/opt/ingress-table/node_modules/bluebird/js/release/timers.js:76:13)
Jun 02 13:35:48 ingress-table node[298]:     at ontimeout (timers.js:386:14)
Jun 02 13:35:48 ingress-table node[298]:     at tryOnTimeout (timers.js:250:5)
Jun 02 13:35:48 ingress-table node[298]:     at Timer.listOnTimeout (timers.js:214:5)
Jun 02 13:35:48 ingress-table systemd[1]: microflo-streetlights.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jun 02 13:35:48 ingress-table systemd[1]: microflo-streetlights.service: Unit entered failed state.
Jun 02 13:35:48 ingress-table systemd[1]: microflo-streetlights.service: Failed with result 'exit-code'.
Jun 02 13:35:49 ingress-table systemd[1]: microflo-streetlights.service: Service hold-off time over, scheduling restart.
Jun 02 13:35:49 ingress-table systemd[1]: Stopped microflo-streetlights.service.
Jun 02 13:35:49 ingress-table systemd[1]: Started microflo-streetlights.service.
Jun 02 13:36:11 ingress-table node[389]: INFO: No runtime id set, generated one: 6573766f-57f9-4e99-83c2-47022b02775c
Jun 02 13:36:11 ingress-table node[389]: Using serial baudrate with /dev/ttyACM1 115200
Jun 02 13:36:12 ingress-table node[389]: Available serial ports:  [ '/dev/ttyACM0', '/dev/ttyACM1', '/dev/ttyAMA0' ]
Jun 02 13:36:12 ingress-table node[389]: Using port: /dev/ttyACM1
Jun 02 13:36:12 ingress-table node[389]: MicroFlo runtime listening at localhost:4444
Jun 02 13:36:14 ingress-table node[389]: Open in Flowhub:
Jun 02 13:36:14 ingress-table node[389]:  http://app.flowhub.io#runtime/endpoint?protocol%3Dwebsocket%26address%3Dws%3A%2F%2Flocalhost%3A4444%26id%3D6573766f-57f9-4e99-83c2-47022b02775c```
```

commit d5425f83 (2017-06-01): Timeout and error if unable to open device communication

Fixes #130
PR #132 (2017-06-01): Timeout and error if unable to open device communication
Fixes #130