microflo/microflo #133: Can't launch node microflo.js runtime

microspace, 2017-06-08:
I get timeout error after trying to launch runtime
[image]
I've uploaded Standalon.ino into Arduino UNO, and connected it to PC

jonnor, 2017-07-11:
Hi @microspace. Do you see the Standalone.ino sketch running on the Arduino? It should be blinking the quickly, around 3 times per second.

Also, can you try the updated instructions for Arduino found here: https://github.com/microflo/microflo-example-arduino ?

jonnor, 2017-07-24:
I tested again on an Arduino Uno and I can reproduce the issue. It happens because the Uno restarts itself when connecting on serial. Working on a fix.

closed 2017-07-24 by jonnor

jonnor, 2017-07-24:
Fixed in 0.4.1

commit 95426ea5 (2017-07-24): runtime: Support waiting before opening communication

This is needed on some Arduino devices that restart
when serial communication is initiated,
like the Arduino Uno.
Defaults to not waiting, pass --wait-timeout 3 to enable

Fixes #133