microflo/microflo #135: grunt rebuild failed

tommyjtl, 2017-06-21:
after adding components i did `grunt build`, but the output was confusing

```
Running "exec:update_components" (exec) task
make: `components' is up to date.

Running "exec:build_runtime" (exec) task
rm -rf build/tests
mkdir -p build/tests
./node_modules/.bin/microflo generate --target linux --library ./components/linux-standard.json graph.fbp build/tests/runtime.cpp
c++ -std=c++0x -o build/tests/runtime build/tests/runtime.cpp -DLINUX -I./node_modules/microflo/microflo -lutil -lrt
>> In file included from build/tests/runtime.cpp:10:
>> In file included from /Users/tommyjtl/Documents/code/playground/microflo/node_modules/microflo/microflo/main.hpp:22:
>> In file included from ./node_modules/microflo/microflo/linux_main.hpp:
>> 7:
>> ./node_modules/microflo/microflo/linux.hpp:22:10: fatal error: 'pty.h' file not found
>> #include <pty.h>
>>          ^
>> 1 error generated.
>> make: * [runtime] Error 1
>> Exited with code: 2.
>> Error executing child process: Error: Command failed: make runtime
>> In file included from build/tests/runtime.cpp:10:
>> In file included from /Users/tommyjtl/Documents/code/playground/microflo/node_modules/microflo/microflo/main.hpp:22:
>> In file included from ./node_modules/microflo/microflo/linux_main.hpp:7:
>> ./node_modules/microflo/microflo/linux.hpp:22:10: fatal error: 'pty.h' file not found
>> #include <pty.h>
>>          ^
>> 1 error generated.
>> make: * [runtime] Error 1
Warning: Task "exec:build_runtime" failed. Use --force to continue.

Aborted due to warnings.
```

is there something wrong with the Grunt setting file?

tommyjtl, 2017-06-21:
@jonnor

jonnor, 2017-07-11:
Hi @tommyjtl. It sounds like there are some dependencies missing for the Linux build. 

Which platform/microcontroller are you developing for?

Since a couple of releases ago, it is no longer neccesary to build or modify anything in `microflo` itself to use custom components. For Arduino, see the updated documentation here: https://github.com/microflo/microflo-example-arduino

closed 2017-09-15 by jonnor