microflo/microflo #137: Task "mochaTest:nodejs" failed

Rollie-Tyler, 2017-07-09:
Hi,

When i am trying to follow the section [''Developing/Hacking"](https://github.com/microflo/microflo#developinghacking), i meet some issues during  ''**npm test**''.

I work on GNU/Linux Ubuntu 17.04.

When i am doing ''**npm install**", i obtain this warning
`npm WARN optional Skipping failed optional dependency /chokidar/fsevents:
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.1.2`
It seem it's not important because fsevents is only for Mac OS-X. But, what is the impact for me ?

And, when i launch ''**npm test**", i obtain this 
```> microflo@0.3.55 test /home/alexandre/Documents/microflo
> grunt test

Running "test" task

Running "build" task

Running "coffee:lib" (coffee) task
>> 12 files created.

Running "coffee:spec" (coffee) task
>> 6 files created.

Running "webpack:lib" (webpack) task
Version: webpack 2.6.1                 
      Asset     Size  Chunks                    Chunk Names
microflo.js  1.17 MB       0  [emitted]  [big]  main

WARNING in ./lib/componentlib.coffee
81:14-31 Critical dependency: the request of a dependency is an expression

WARNING in ./lib/componentlib.coffee
82:17-42 Critical dependency: the request of a dependency is an expression

WARNING in ./lib/runtime.coffee
782:10-23 Critical dependency: the request of a dependency is an expression

WARNING in ./lib/runtime.coffee
792:33-53 Critical dependency: the request of a dependency is an expression

WARNING in ./~/bindings/bindings.js
76:22-40 Critical dependency: the request of a dependency is an expression

WARNING in ./~/bindings/bindings.js
76:43-53 Critical dependency: the request of a dependency is an expression

Running "copy:browserdeps" (copy) task

Running "mochaTest:nodejs" (mochaTest) task

  1) "before all" hook
  2) "after all" hook

  0 passing (2s)
  2 failing

  1)  "before all" hook:
     Uncaught AssertionError: expected [Error: Subprocess wrote on stderr: 'test/start-runtime.sh: 8: 
test/start-runtime.sh: ./build/linux/firmware: not found
'] to not exist
      at Object.exports.noError (node_modules/fbp-spec/src/expectation.coffee:80:31)
      at node_modules/fbp-spec/src/mocha.coffee:66:19
      at Socket.<anonymous> (node_modules/fbp-spec/src/subprocess.coffee:45:14)
      at readableAddChunk (_stream_readable.js:153:18)
      at Socket.Readable.push (_stream_readable.js:111:10)
      at Pipe.onread (net.js:537:20)
  
  2)  "after all" hook:
     Error: Timeout of 2000ms exceeded. For async tests and hooks, ensure "done()" is called; if returning a 
Promise, ensure it resolves.
  
Warning: Task "mochaTest:nodejs" failed. Use --force to continue.

Aborted due to warnings.
npm ERR! Test failed.  See above for more details.
```
If i well understand, firmware that i must found under /build/linux, is not build. No ?
Could you enlighten me?

PS : I am not very familiar with node.js but i try to understand.

jonnor, 2017-07-11:
Hi @Rollie-Tyler. Are you using MicroFlo to program an Arduino? If so you no longer need to build or modify `microflo` itself. Updated documentation can be found here: https://github.com/microflo/microflo-example-arduino

Rollie-Tyler, 2017-07-11:
Hi @jonnor .
Yes, for now, i'm using microflo to program an arduino to understand how works microflo.
My goal is to add some news targets like microchip microcontroller or nuvoton processor. I use some microchip microcontroller  many year and i would like use microflo for some specifics part where flow diagrams are more efficiency than write this in C.
I will try your updated documentation and make you a return.

jonnor, 2017-07-24:
Your problem with the Arduino Uno should have been fixed now, see comments in https://github.com/microflo/microflo-example-arduino/issues/1

There is a little bit of documentation on how to port MicroFlo to new devices linked in https://github.com/microflo/microflo#extending
Feel free to open issues about any questions you have around that.

closed 2017-07-24 by jonnor